omarcostahamido/PVM #5: Going out of sync!

omarcostahamido, 2022-10-05:
under some circumstance devices, that have been triggered to start at the same time, go out of sync.
@gggfffsss was this still happening with the latest version?
This issue needs to be further investigated... and, it is still happening, it needs to be fixed.

omarcostahamido, 2022-10-14:
Issue #11 is focusing on the `buffer deadlock` problem which might be what causes the videos to go out of sync.

omarcostahamido, 2022-10-25:
@SSSSam141 I know you focused on the `buffer deadlock` issue, which could be the reason for things going out of sync. In any case, I would like to ask you to confirm that synced videos is really no longer an issue and that omxplayer has solved it all. Best way to confirm that could be with the newly added [examples](https://github.com/omarcostahamido/PVM/tree/main/examples), I would suggest 😉 
If all is good in this domain then this issue could be closed.

KaneBetter, 2022-10-31:
Find some projects for sync omxplayer.
https://github.com/turingmachine/omxplayer-sync
https://github.com/bmoren/node-omxplayer-sync

omarcostahamido, 2022-10-31:
woah, thank you for finding those @KaneBetter !
I'm boosting this issue to `High`. I'd love to see this being closed next, before the dual screen issue which will bring added complexity here.
take it away!

KaneBetter, 2022-11-01:
> woah, thank you for finding those @KaneBetter ! I'm boosting this issue to `HIGH`. I'd love to see this being closed next, before the dual screen issue which will bring added complexity here. take it away!

We still need to reproduce this issue first. Maybe this issue doesn't exist in OMXPlayer.
If this issue does exist, we have two options to choose from.
One is to send a timestamp with the start function.
Two is to let each Raspberry Pi talk to the other.

KaneBetter, 2022-11-03:
It seems omxplayer and python processes themselves run smoothly in each RPI. The average time for the system to execute the command is 70ms.
We also observed this issue #35  in our tests. So we think bringing this #35  up is an excellent solution.

KaneBetter, 2022-11-17:
#38 fixes this.

closed 2022-11-17 by KaneBetter