omarcostahamido/PVM #10: Add requirements.txt from `pip freeze` for easier installation

rustberry, 2022-10-08:
Low priority for now.

omarcostahamido, 2022-10-08:
Yes, I like that, even though it is pretty short list of dependencies.
Can you include in the todo list also updating the readme? Note: even if the instructions are only to now include the installation command with the txt, can you please still reference the python dependencies? Maybe this means updating the first section.

omarcostahamido, 2022-10-11:
@rustberry I've just updated your OP with a short TODO list.
This one is pretty straight forward. Since you already have 2 other issues assigned, maybe @KaneBetter would like to take this issue as a _warmup_ until he finds an issue he would like to focus on?

omarcostahamido, 2022-10-12:
this was fixed by #14

closed 2022-10-12 by omarcostahamido