omarcostahamido/PVM #11: Investigate buffer deadlock prevented issues.

SSSSam141, 2022-10-08:
When playing jellyfish video, it will pause by the issue of buffer deadlock prevented for serveral times. 

- [ ] Figure out what happens when buffer deadlock prevented protection occurs.
- [ ] Decrease the occurrence of buffer deadlock prevented.

omarcostahamido, 2022-10-08:
Thank you @SSSSam141 for opening this issue. And so it begins... :)
Fyi, the website where you download that jellyfish sample video includes different compression renders that you may also try in your experiments. 
Good luck, and please keep us posted.

omarcostahamido, 2022-10-11:
@SSSSam141 please feel free to share here some the things you've been trying.

KaneBetter, 2022-10-12:
Just find out that if you run `sudo apt install vlc` in the terminal.
After that, the error message disappeared.
However, this still needs to be approved by someone else.

SSSSam141, 2022-10-12:
I have tried to run vlc to play the video: jellyfish without using Max. The problem was still there so we can make sure that the cause lies in the part of VLC. I have checked that our Raspeberry pi model is 4 (cmd: `cat /proc/device-tree/model` ), its system is  bullseye 11 (cmd: `cat /etc/os-release`), and the vlc version is 3.0.16 (cmd: `vlc -v`).  
Also, we can use debug mode for the vlc, change line 6 at pvm.py file: `inst = vlc.Instance('--verbose 2')`, which will be more convenient for getting more information when debugging.

I tried to upgrade the version of VLC but the buffer deadlock prevented message is still there. 
The completely remove and reinstall VLC commands: In a terminal, run:
`sudo apt-get remove --purge vlc`
`sudo apt-get autoremove`
`sudo apt-get autoclean`
`sudo apt-get install vlc`
Done. 
Thus, reinstall the vlc with the latest version does not help to solve the problem.

The error message: `mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module` happens earlier than the buffer deadlock prevent issue #12 and the buffer deadlock prevent is an error of main decoder error. The mmal_vout is the decoder of vlc which failed to open, so that I think we should solve the problem of #12 first then we can see whether this problem is still there. I will move to see #12 first.

omarcostahamido, 2022-10-13:
> I have tried to run vlc to play the video: jellyfish without using Max. The problem was still there so we can make sure that the cause lies in the part of VLC.

huh... that's strange.

> I have tried to run vlc to play the video: jellyfish without using Max. The problem was still there so we can make sure that the cause lies in the part of VLC.

Nice.

> Thus, reinstall the vlc with the latest version does not help to solve the problem.

Thx for testing this. Please remind me: is bullseye 11 the latest raspberrypiOS?

> The error message: mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module happens earlier than the buffer deadlock prevent issue [...]

Perfect! Prioritizing :) Feel free to change the label on #12 to `High`. Also, add it here as a step on the TODO list.

SSSSam141, 2022-10-13:
@omarcostahamido Bullseye 11 is the latest version for raspberryPiOS.

SSSSam141, 2022-10-14:
After I have reinstalled the os with the old version buster 10, the problem still exists. I will try to find the cause now.

omarcostahamido, 2022-10-23:
@SSSSam141 any updates on this?

SSSSam141, 2022-10-24:
> @SSSSam141 any updates on this?

The new media player: OMXPlayer does not have this issue. I think we can close this issue.

omarcostahamido, 2022-10-25:
Then I suggest closing this issues with the PR that makes that transition too! #19

PR #19 (2022-10-25): Replace VLC with OMXPlayer
Close #11
Fix #16 
- Added pvm_omx.py
- Added build script for OMXPlayer
- Added TODO list in the code

Future scope of work:

- [ ] revise README.md with current implementation
- [ ] finish all TODO list
closed 2022-10-25 by KaneBetter