omarcostahamido/PVM #12: Failed to open Xsplitter:mmal_vout module

xrk-git, 2022-10-11:
Potential solution: https://github.com/RPi-Distro/vlc/issues/30

omarcostahamido, 2022-10-11:
Hi @xrk-git . Thank you for opening this issue. A little more context would be great. When does it happen/what are the steps to reproduce? (and also, what is it related to?)

xrk-git, 2022-10-11:
@omarcostahamido Thanks for your suggestions. I will revise that later :)

SSSSam141, 2022-10-12:
In fact, only me and Ruokun has encounterd this problem, while Kaiqin and Zejin did not happen on their machines. The way
to reproduce problem is to use Max to play the jellyfish video.

- go to the directory of PVM
- `source PVM/bin/activate`
- `python pym.py`
- Go to max to open the pvm.maxpat
- click the reset button and then click the start button.

Then you can check the error message from the terminal of Raspberry.

SSSSam141, 2022-10-12:
I will try to find a way to configure the decoder for vlc or substitude the original decoder with another feasible one, but I am not sure it would work or not.

omarcostahamido, 2022-10-13:
> @omarcostahamido In fact, only me and Ruokun has encounterd this problem, while Kaiqin and Zejin did not happen on their machines. The way to reproduce problem is to use Max to play the jellyfish video.

@SSSSam141 just to be clear: 
i) you are all using a mac computer with Max as the controlling machine, and different raspberry pi devices loaned by @gggfffsss as the playing device? I'm trying to understand if `their machines` is referring to the raspberry pi device or the controlling machine.
ii) the error only appears after you click either the `start` message, or the toggle next to it, on the `pvm.maxpat`. But even when the error appears the video still starts playing.

My educated guess, and I think yours as well, is that Max is not the culprit here.
I wonder if the problem is consistent. And if you want to try swapping your devices, just to confirm that the problem is not originating on the controller machine.

SSSSam141, 2022-10-13:
> @SSSSam141 just to be clear: i) you are all using a mac computer with Max as the controlling machine, and different raspberry pi devices loaned by @gggfffsss as the playing device? I'm trying to [...]

i) `their machines` is referring to the raspberry pi device. We are using our own laptop to be the control machine currently.
ii) The error occurs when I click start and then wait until the video pauses itself. The video will firstly play normally for a few seconds. When the error occurs, it continues to play until the buffer deadlock prevented issues, it pauses.

I think this issue should not come from the controller machine and I will try to verify that later.

SSSSam141, 2022-10-13:
@omarcostahamido 
Thanks to Kaiqin @KaneBetter for providing the link to the discussion: `https://raspberrypi.stackexchange.com/questions/133541/vlc-not-hardware-accelerated-on-raspberry-pi-400`
It is said that the MMAL module, which is needed by VLC, is now deprecated on Raspi 4. I want to try to use another version of RaspberryPiOS version (buster 10) to do that. But I need to reinstall the OS on the SD card. Do we have a SD card reader?

omarcostahamido, 2022-10-13:
Thx. The answer is yes. @gggfffsss will give you access to it.

SSSSam141, 2022-10-14:
@omarcostahamido 
I have reinstalled the os on Raspeberry with the version of buster 10 and the mmal_vout module can be turned on normally and there is no tearing problem while the buffer deadlock still exists. This issued proves that the bullseye 11 does not support the mmal_vout and I think we can close this now. I will turn to the main decoder error: buffer deadlock prevented issue #11.

closed 2022-10-14 by SSSSam141

omarcostahamido, 2022-10-14:
perfect. thank you @SSSSam141