omarcostahamido/PVM #13: Burn a brand-new rpi image on the micro SD card and set up the project from scratch, to make sure we've covered all the needed dependencies.

rustberry, 2022-10-11:


omarcostahamido, 2022-10-11:
@rustberry I just simplified the issue name, and move the longer version into the issue OP - thx for creating it.

if anything else, it might be a good to add new and updated versions of the OS so that we might benefit from relevant updates that might have come out in the meantime.

omarcostahamido, 2022-10-14:
@rustberry can we consider this issued mostly fixed given the work reported in #12 ?
I would still suggest that this issue could be closed with the addition on the README of the complete instructions on how to install and setup the proper OS version. Can you do that and create a PR including the message `this fixes #13`?

omarcostahamido, 2022-10-17:
_bump_

just looking around the high priority issues and trying to figure out when they'll be closed.

rustberry, 2022-10-18:
Yeah, I think this could be closed soon. After we finally decided to use OXMPlayer I will update the related instructions.

SSSSam141, 2022-10-18:
Done.

omarcostahamido, 2022-10-19:
@SSSSam141 could you please close this issue with the PR/commit that contains the added instructions?

omarcostahamido, 2022-10-20:
this was closed by #17

closed 2022-10-20 by omarcostahamido