omarcostahamido/PVM #16: Looking for an alternative media player

KaneBetter, 2022-10-15:
Since in order to solve the decode  @SSSSam141 found that he needed to roll back the system version of Raspberry, I tried to solve the problem with another idea. At the same time, since this problem will involve the problem of unsynchronized, frame tearing afterward, so I set the level of this issue to high
- [ ] OMXPlayer
- [ ] Kodi
- [ ] Plex Media Player
- [ ] OSMC
- [ ] XBian
- [ ] LibreELEC
- [ ] RasPlex

omarcostahamido, 2022-10-15:
Thanks for starting this issue @KaneBetter .
It seems that this could be a parent issue for multiple issues (and branches) testing all those different options.
If you need a simpler `pvm.maxpat` for quick testing purposes, let me know. Else, as you've realized by now, you can also quickly draft an osc python script to send a single command.

Speaking of which, I don't want to close the door on the controller patch (`pvm.maxpat`) to use some other protocol other than OSC. For example, UDP (which OSC is based on), TCP, or even http requests, and websockets! I doubt we'll get significantly better speed with those but, again, I'm happy to do some max patching to help test that.

KaneBetter, 2022-10-15:
> Thanks for starting this issue @KaneBetter . It seems that this could be a parent issue for multiple issues (and branches) testing all those different options. If you need a simpler `pvm.maxpat` for [...]

Thanks for reminding me of this, we use Zenhub to manage our projects, it works as a plugin for Github, easy to create Epic, Sprint and manage the progress of issues, Roadmap etc.
I will send the invite link to you and Grant

rustberry, 2022-10-16:
Also tried compiling VLC from the source but no luck. (https://thepi.io/how-to-compile-vlc-media-player-with-hardware-acceleration-for-the-raspberry-pi/ https://www.techybugz.com/raspberry-pi/raspberry-pi-vlc/)

Some needed libraries cannot be found on Rabian OS (at least bullseye 11). Trying to compile without them resulted in errors during `make`.

KaneBetter, 2022-10-17:
After trying other players and play centers, I found that OMXPlayer runs smoothly in the buster(10) system environment and is a more suitable solution.
But this still has the bug that when you loop the video, you can only set the rate for one of the loops
The following are two solutions.
- [ ] Modify the source code of OXMPlayer Wrapper:https://github.com/willprice/python-omxplayer-wrapper/blob/master/omxplayer/player.py
- [ ] Change our own code

omarcostahamido, 2022-10-17:
@rustberry have you tried compiling VLC on bullseye 10?

@KaneBetter thank you so much for trying all those players!
It doesn't surprise me that omxplayer runs smooth - it is done in `c++` right?

> But this still has the bug that when you loop the video, you can only set the rate for one of the loops

I'm not sure I understood it. Do you mean that if you set a playback rate other than normal speed, when it loops and starts from the beginning it changes the playback speed back to normal? That sounds like an omx bug... 
From the 2 solutions you present, does the second on imply having the control patch sending a `change playback speed` command everytime the video clip loops? (that would means that it must keep track of playback position).

My question about omxplayer now would be more in the sense of: what are the remotely accessible commands, and which protocol is it compatible with. If no networking protocol is in place, can we start planning on how to implement it? 😃

KaneBetter, 2022-10-18:
> @rustberry have you tried compiling VLC on bullseye 10?
>
> @KaneBetter thank you so much for trying all those players! It doesn't surprise me that omxplayer runs smooth - it is done in `c++` right? [...]

Your understanding is correct, the python program controls the OMXPlayer by using the OMXPlayer Wrapper, and I have uploaded a branch at https://github.com/omarcostahamido/PVM/tree/UseOmxplayer.

rustberry, 2022-10-18:
> @omarcostahamido: @rustberry have you tried compiling VLC on bullseye 10?

Yep, after downgrading to bullseye 10 also tried to compile vlc on 10. Same issue: cannot install all the required packages for compiling 😞. Though the bootstrapping and configuring stages went okay without showing errors, the final compiling stage would fail. 
[image]

Compared to 11 bullseye 10 has more packages satisfied, but still not enough.

Try to install manually some of the missing packages, but they all seemed to be removed for being outdated.
Some examples:
https://discourse.libsdl.org/t/raspberry-pi-4-support-gles-1-1/26350
https://forums.raspberrypi.com/viewtopic.php?t=244437#p1490886

KaneBetter, 2022-10-18:
Find the same issue in OMXPlayer: https://github.com/popcornmix/omxplayer/issues/799

KaneBetter, 2022-10-18:
Find the solution for our case in a Pull request. https://github.com/popcornmix/omxplayer/pull/678
However, the developers don't seem to want to merge.

KaneBetter, 2022-10-19:
We successfully added the option for OMXPlayer to play videos at variable speed, and created the build shell script, see commit: https://github.com/omarcostahamido/PVM/commit/c3c57b931bb6e1b9fc8ffbe6490c0cc1187d302a

KaneBetter, 2022-10-19:
Successfully called force-fps in pvm_omx.py to implement variable rate, see commit: https://github.com/omarcostahamido/PVM/commit/d4b12b9025059062df6c540d93dc09d29ffa42b4
So far, all the basic functions have been implemented, continue to modify the code format

SSSSam141, 2022-10-19:
I think we have successfully find an alternative to VLC that is OMXPlayer. While I have found that the current logic in `pvm_omx.py` still has some problem, when I try to press twice reset, it starts to play video. There is no similar instance variable in OMXPlayer. So I think we need to focus on making the logic work for the control pad without bugs. Therefore, this issue could be closed and turn to next issue for implementing `pvm_omx.py`.

SSSSam141, 2022-10-19:
> 

@omarcostahamido I recommend you to input the command line by line since there might be some cases that the script does not cover.

KaneBetter, 2022-10-19:
> trying to build @KaneBetter 's version of omx, on a vbox with raspios buster 10: [image] any ideas?

Can you try input `./build_omxplayer.sh`

rustberry, 2022-10-19:
> Can you try input ./build_omxplayer.sh

@omarcostahamido This worked for me. Also if you try `bash build_omxplayer.sh` it would also work. Think we also need to add something to the script, like `apt purge omxplayer` to remove any possible existing installation.

To test the new binary, run `omxplayer --force-fps 10 jellyfish720p.mp4` which would play the video at the forced rate.

omarcostahamido, 2022-10-19:
right, so, at the end of the building process it says that some packages are missing

[image]

rustberry, 2022-10-19:
How about running these before running the script again:
`sudo apt purge omxplayer && sudo apt-get update --allow-releaseinfo-change`
I didn't run into this, strange 🤔 

ref:
https://www.linuxtutorials.org/This-must-be-accepted-explicitly-before-updates-for-this-repository-can-be-applied-See-apt-secure-8-manpage-for-details/

SSSSam141, 2022-10-20:
> on my end at least `omxplayer` command is not recognized
> 
> following what you said: [image] [image]
> 
> I am running this version on my vbox: 2021-01-11-raspios-buster-i386 [[1]](https://downloads.raspberrypi.org/rpd_x86/images/) This is called raspberry pi for mac/windows desktop [[2]](https://www.raspberrypi.com/software/raspberry-pi-desktop/) Maybe that is enough to drive the difference we are noticing here? Could you please clarify which precise image version of the OS you're using?
> 
> This is related to issue #6 : we need very clear instructions on the readme about which version to use and how to get it.

@omarcostahamido I have updated the readme file regarding the installation of new Raspberry Pi version on my branch. Please check. If you have any question, please let me know.

KaneBetter, 2022-10-20:
> on my end at least `omxplayer` command is not recognized
>
> following what you said: [image] [image]
>
> I am running this version on my vbox: 2021-01-11-raspios-buster-i386 [...]

The reason you are getting this error is that the statement you ran made you delete the OMXPlayer, the correct action is as follows.
`sudo apt purge omxplayer && sudo apt-get update --allow-releaseinfo-change`
`cd to folder`
`./build_omxplayer.sh`
`omxplayer --force-fps jellyfish720p.mp4`

KaneBetter, 2022-10-21:
Since we have found the OMXPlayer, I am going to make PR to merge UseOmxplayer branch into main.
So that other developer can create their own branch based on this.

omarcostahamido, 2022-10-22:
@KaneBetter thank you for creating the PR with the proper message that will automatically close this issue.

omarcostahamido, 2022-10-23:
> next up for me: trying to setup the same version of raspios as the one listed on the readme now.

I can't actually create a virtual machine to install the same version of the OS that you're using because... rpi is an arm-based hardware. There might be a way to emulate it [[1](https://joshondesign.com/2021/04/15/emu_pi_mac)] but it'll be inefficient. I will try to install a newer version of the raspios just for the sake of it and report back.

omarcostahamido, 2022-10-24:
> I will try to install a newer version of the raspios just for the sake of it and report back.

after a little bit of fiddling around I was able to get a buster11 install running and apparently reaching a little further this time around but without complete success.
[image]

(though less than desirable, It is not absolutely essential that I am able to run it on my virtualbox. We need it to run _really_ well on those rpi and flatscreens!)

please soldier on.

KaneBetter, 2022-10-25:
> 

Yeah, it is definitely going further. I look up those packages, and I found [this](https://github.com/RetroPie/RetroPie-Setup/issues/2509). 
These packages are used for building the binary and these packages can't be found on every platform RPI support, see [here](https://github.com/RetroPie/RetroPie-Setup/issues/2508).
So the reason why I didn't come across this is that we burn our image through RPI imager which automatically installs these packages.

closed 2022-10-25 by KaneBetter

omarcostahamido, 2022-10-25:
@KaneBetter do you reckon a new issue should be open to `Improve the build shell script` then?

PR #19 (2022-10-25): Replace VLC with OMXPlayer
Close #11
Fix #16 
- Added pvm_omx.py
- Added build script for OMXPlayer
- Added TODO list in the code

Future scope of work:

- [ ] revise README.md with current implementation
- [ ] finish all TODO list