omarcostahamido/PVM #18: Need help with Continuous Deployment script

KaneBetter, 2022-10-20:
I have pushed a new branch called AddCDscript: https://github.com/omarcostahamido/PVM/tree/AddCDscript
The importance of this script, which allows for automatic deployment and saves us a lot of time.
Because our test environment has many rpis, when we modify our code, we need to deploy to each rpi and then test, manually ssh to each deployment will waste a lot of time.
So I create this issue and branch to help add this Continuous Deployment script.
I have already written some code in this file: https://github.com/omarcostahamido/PVM/blob/AddCDscript/deploy_code_to_rpi.sh.
Unfortunately, my system was upgraded to macOS13 and does not support pdsh installation, so hopefully, someone can take over this script.
The functionality I expect this script to achieve is as follows:

- ssh into each rpi
- kill the running python process
- git fetch the latest version or just delete and redownload the repo(The details are up to you)
- run the python process again

omarcostahamido, 2022-10-22:
interesting idea.
(I wonder if the script could check the versions and continue re-deployment only if it is outdated - maybe something for later / not a priority).
This works assuming that all the devices that we want to update/re-deploy are listed on `max-init.txt`. (this should be clarified somewhere - perhaps on the readme during the section explaining this new script?)

KaneBetter, 2022-10-25:
> interesting idea. (I wonder if the script could check the versions and continue re-deployment only if it is outdated - maybe something for later / not a priority). This works assuming that all the devices that we want to update/re-deploy are listed on `max-init.txt`. (this should be clarified somewhere - perhaps on the readme during the section explaining this new script?). What happens if it doesn't find the devices on that list?

Yes, the initial thought is that the Control PC will run this script and the new code can be deployed on each RPI listed on the `max-init.txt`.
Regarding checking the versions, although there are many ways to do that, in our case, I don't think it will be a good idea.
Because this kind of update must be done periodically, and here we want the immediate effect.
And we definitely need to explain how the script works in the README.md
For the question `What happens if it doesn't find the devices on that list?`, the `[pdsh](https://linux.die.net/man/1/pdsh)` command will just ignore those hosts.

omarcostahamido, 2022-10-25:
thx. LGTM

PR #26 (2022-10-30): Add deploy_code_to_rpi.sh script
Fix #18 
Might Fix #20 
 I suggest moving videos into `/home/pi/Videos/` for autostart use.
The deploy_code_to_rpi.sh can extract the host IP from max-init.txt file and then use pdsh command ssh into each host and then run the command mentioned in README.md to deploy code.
closed 2022-10-30 by KaneBetter