omarcostahamido/PVM #35: Timestamp for Max8

KaneBetter, 2022-11-03:
We test two RPIs in the classroom.
Found out that Max8 command for each rpi is not at the same time.
There is a 0.5s delay between two rpis.

We want Max8 can send out commands like this: start 18:00:00(current time).
You can see the video in slack https://swe276pcapsto-5ba9154.slack.com/files/U043JMJPPQE/F0490JZ97GF/img_0585.mov

KaneBetter, 2022-11-05:
We have confirmed that MAX does run pvm_send at the same time.
[image]
So the cause of the delay must be inside the UDP send process.

KaneBetter, 2022-11-05:
With this new pvm_send, our solution will become very simple, send the current time by UDP, then all RPI receive the time and postpone the operation backward for 3s to solve the synchronization problem perfectly.

KaneBetter, 2022-11-05:
Pushed branch https://github.com/omarcostahamido/PVM/tree/AddTimeInUDP
TODO: rewrite the python script to parse the time.

omarcostahamido, 2022-11-12:
> There is a 0.5s delay between two rpis.

Oh no...

> We have confirmed that MAX does run pvm_send at the same time.
So the cause of the delay must be inside the UDP send process.

Creating a new UDP object would be quite the achievement, not impossible though. I've done one [here](https://github.com/omarcostahamido/One-Quick-Solution/tree/master/MaxMsp/och.udp). Never tested against the default one in max for performance though. I built it in order to easily receive the IP+port of the sender in the received end.

> With this new pvm_send, our solution will become very simple, send the current time by UDP, then all RPI receive the time and postpone the operation backward for 3s to solve the synchronization problem perfectly.

Oh interesting... you know, this is now starting to remind me more about [Ableton Link](https://github.com/Ableton/link) that is a technology aimed at solving this exact problem of synchroization across different audio applications, and Max has a [Link package](https://cycling74.com/feature/abletonlink) too! (which I never really got much deep into it). Food for thought, or just another possible elegant solution?

> Pushed branch

will check this out

PR #38 (2022-11-17): Add timestamps to `pvm_send` abstraction and `pvm.py`
This fix #35 and #5
closed 2022-11-17 by KaneBetter