omarcostahamido/PVM #42: 2 monitors are not synchronized

KaneBetter, 2022-11-10:
When we test two videos' python script, we found out that r=press start once and then pause and then press start again has a certain chance of causing the two monitors to be out of sync.

TODO: Reproduce the bug first, then determine why the bug occurred

SSSSam141, 2022-11-10:
The 2 media.position functions are totally 0.03s ~ 0.04s, at most time the offset of the 2 videos are under this range and we can consider them synchronized while as the video continue to loop, the offset of them becomes larger. I will investigate more into this problem and try to find out what is the cause of this problem.

SSSSam141, 2022-11-11:
After checking the source code of set_position() function, it is "Set the video to playback position to `position` seconds from the start of the video", which should set to the past time not the future time. Then I change the logic for media 1 and media 2, only set the media which is playing relatively faster back to the position where media is playing relatively slower. Then now the 2 videos are synchronized.

omarcostahamido, 2022-11-12:
> Then I change the logic for media 1 and media 2, only set the media which is playing relatively faster back to the position where media is playing relatively slower. Then now the 2 videos are synchronized.

🐢 🐇

KaneBetter, 2022-11-21:
This is not the case with the latest version.

closed 2022-11-21 by KaneBetter

omarcostahamido, 2022-11-21:
good to hear!