omarcostahamido/PVM #43: Frame setting feature implementation

xrk-git, 2022-11-10:
Acquire the playing video's frame rate and set the Omxplayer exhibition frame rate.

omarcostahamido, 2022-11-12:
@xrk-git where/how did you complete/fix this?
could you please use the keywords `this fixes #43` on the PR that fixes this before merge? That way it will create a link here and close this with that PR. This saves you time to manually close an issue, AND creates a link that makes it easier to go back and understand where exactly were the changes that fixed it. (you can still go there and write `this refers to #43` just to create that link)
thank you

xrk-git, 2022-11-15:
This fixes #47

omarcostahamido, 2022-11-15:
@xrk-git the PR fixes the issue, not the other way around.
I'm going to edit the OP on #47 for you to include those words and you'll see the difference 😉

PR #48 (2022-11-17): set frame rate feature
Fix #43
closed 2022-11-17 by xrk-git