omarcostahamido/PVM #49: Use case clarify for two video

KaneBetter, 2022-11-17:
We should clarify our two video control use cases first and then write example UDP commands here.
At the same time, #30 will change according to this issue.

KaneBetter, 2022-11-17:
User input: file 1.mp4
Log output: You need to input 2 filenames.
Or
User input: file 1.mp4
Log output: We consider 1.mp4|1.mp4

KaneBetter, 2022-11-17:
Use case:
User input: rate 2
Log output: display2 rate 2, display7 rate 2

User input: rate 2
Log output: display2 rate 2

User input: rate 2|
Log output: display2 rate 2

User input: rate 2|
Log output: display2 rate 2, display7 rate 2

KaneBetter, 2022-11-17:
Chose log ouput for 

- rate 2
- rate 2|
- file 1.mp4
- file 1.mp4|

KaneBetter, 2022-11-17:
@omarcostahamido Could you take a look at the command above?

omarcostahamido, 2022-11-17:
Hi. So my first question would be: can we detect if there is only 1 screen connected or 2?
It is not worth it to launch 2 video windows if the device only has 1 display connected.

If the device has 2 displays connected I don't see any advantage in not sending any video through both displays.
In this case, it would be easy to say that if the user is only sending control information for 1 display then that should be replicated on the other screen - as in loading the same video, performing the same offsets etc.

E.g. `/PVM HH MM SS file jelly.mp4` > will load the `jelly.mp4` file on both display 2 and display 7

For sending control information to both displays, we can follow your notation and use `|`.

E.g. `/PVM HH MM SS file jelly.mp4|zoe.mp4` > will load `jelly.mp4` on display 2 and `zoe.mp4` on display 7

However, there may come some situations when the user might want to just send control information to one display and not the other. In this situation maybe we need to come up with an agreed symbol that will just not do any changes to that display. Let's say we use the symbol `=` - I'm open to suggestions.

E.g. `/PVM HH MM SS file =|sunset.mp4` > will keep display 2 as it is and replace file on display 7 with `sunset.mp4`
E.g. `/PVM HH MM SS set_rate 1.5|=` > will keep display 7 as it is and change the rate on display 2 to `1.5`

How does that sound?

omarcostahamido, 2022-11-17:
> _It is not worth it to launch 2 video windows if the device only has 1 display connected.
If the device has 2 displays connected I don't see any advantage in not sending any video through both displays._

Question: does omxplayer launches a black window if it is executed without a file defined? My guess it that it won't launch any window since it won't even know what size to give that window in the absence of any media.

> _In this case, it would be easy to say that if the user is only sending control information for 1 display then that should be replicated on the other screen - as in loading the same video, performing the same offsets etc._

This will also make much more easier to use if the user just wants to have the same video going through the same exact changes without having to manually duplicate the commands everytime. In fact, even if we have different videos going on this will be useful. Imagine a scenario with display 2 playing `fall.mp4` and display 7 playing `spring.mp4`; we will be able to send `/PVM HH MM SS pause` to pause both displays at the same time instead of having to send `/PVM HH MM SS pause|pause`.

KaneBetter, 2022-11-17:
> Hi. So my first question would be: can we detect if there is only 1 screen connected or 2?
> It is not worth it to launch 2 video windows if the device only has 1 display connected.

Raspberry pi can't know how many displays it is connected to.

KaneBetter, 2022-11-17:
> E.g. `/PVM HH MM SS file =|sunset.mp4` > will keep display 2 as it is and replace file on display 7 with `sunset.mp4`
> E.g. `/PVM HH MM SS set_rate 1.5|=` > will keep display 7 as it is and change the rate on display 2 to `1.5`

That's achievable, let me update our use case to include that.

KaneBetter, 2022-11-17:
> This will also make much more easier to use if the user just wants to have the same video going through the same exact changes without having to manually duplicate the commands everytime. In fact, [...]

agreed, let me include this also.

omarcostahamido, 2022-11-17:
Commeting the updates to OP: user may send integer but RPI will see it as floats, like 1.0|2.0.
Right?

Also, all this discussion needs to be well articulated in the readme documentation.

Let me know when (and on which branch) do you want me to try creating a new example Max patch. I can easily help with that. 

Thanks

KaneBetter, 2022-11-18:
We have updated our python logic in the branch TwoVideoControl. @omarcostahamido Could you please add or change the maxpat file in that branch according to the use case we discussed.

omarcostahamido, 2022-11-19:
DONE https://github.com/omarcostahamido/PVM/pull/31/commits/4d2925cf66d0eb3868c860091b689332b5a9c65c Good luck on finalizing `pvm.py`. I made a thorough review of that and noticed a lot of things... 28 review notes. I am still waiting to hear about how it actually feels when working with any of the examples.
And, of course, I'm very much looking forward to see a [release](https://github.com/omarcostahamido/PVM/releases) happening.

KaneBetter, 2022-11-21:
No longer needed.

closed 2022-11-21 by KaneBetter