omarcostahamido/PVM #57: Readme updates

omarcostahamido, 2022-11-27:
The readme needs a thorough revision including:

- [ ] #55 
- [ ] remove https://github.com/omarcostahamido/PVM#known-limitations section
- [ ] add acknowledgements section
- [ ] update https://github.com/omarcostahamido/PVM#structure section
- [ ] on the examples section the part that starts with `Given the objective of making this project available ...` is not from there. maybe we can move this explanation (which needs to be updated to reflect the new available commands and explain their difference and types of values) to the wiki and link it.
- [ ] review https://github.com/omarcostahamido/PVM#helpful-links section

feel free to edit this OP to add more and or transform some of them in other issues if need be.

KaneBetter, 2022-11-27:
I will create a branch called update README.md to solve 2-5 checklists.

KaneBetter, 2022-11-27:
> * remove https://github.com/omarcostahamido/PVM#known-limitations section

Mark this resolved with https://github.com/omarcostahamido/PVM/commit/4af76d636948115875cf04485b4e1d23b5d5cc3b

omarcostahamido, 2022-11-27:
@KaneBetter thx.
I can help with the acknowledgements section once you get things rolling. Just ping me when I can add to it.

KaneBetter, 2022-11-27:
> * on the examples section the part that starts with Given the objective of making this project available ... is not from there. maybe we can move this explanation (which needs to be updated to [...]

Mark this resolved, 
see commit: https://github.com/omarcostahamido/PVM/commit/3cb7f72cf9ca4c46accb7b6c9b39e146cda40687
see wiki: https://github.com/omarcostahamido/PVM/wiki

KaneBetter, 2022-11-27:
> * review https://github.com/omarcostahamido/PVM#helpful-links section

Mark this resolve, see commit https://github.com/omarcostahamido/PVM/commit/3a5f9e4ea1aeb1a75a3aa6aedeb3325ee0b8197c.

KaneBetter, 2022-11-27:
> * update https://github.com/omarcostahamido/PVM#structure section

Mark this resolve, see here https://github.com/omarcostahamido/PVM/tree/UpdateREADME

KaneBetter, 2022-11-27:
> @KaneBetter thx. I can help with the acknowledgments section once you get things rolling. Just ping me when I can add to it.

@omarcostahamido , yes, please.

KaneBetter, 2022-11-28:
> * maybe the https://github.com/omarcostahamido/PVM#raspberry-pi-os section could also go to wiki?

Move this section into wiki
https://github.com/omarcostahamido/PVM/commit/1aef34a4e8112fc97837ebb51a7476d45709423d

omarcostahamido, 2022-11-28:
> > @KaneBetter thx. I can help with the acknowledgments section once you get things rolling. Just ping me when I can add to it.
> 
> @omarcostahamido , yes, please.

DONE in https://github.com/omarcostahamido/PVM/commit/2ffb00953bfe7d302cb901a469409c68e982d5f1

KaneBetter, 2022-11-28:
> * add @KaneBetter's omx_player and Raspberry Pi to the dependencies section?

Mark resolve.
https://github.com/omarcostahamido/PVM/commit/8bc181e621bc4a0df96b795bfc7d0f2f372b730e

KaneBetter, 2022-11-28:
> * review/(simplify?) the installation section

Done here. https://github.com/omarcostahamido/PVM/commit/39ce0759537e8ef56789babdb1fdc450eedc748a

omarcostahamido, 2022-11-28:
> > * on the examples section the part that starts with Given the objective of making this project available ... is not from there. maybe we can move this explanation (which needs to be updated to reflect the new available commands and explain their difference and types of values) to the wiki and link it.
> 
> Mark this resolved, see commit: [3cb7f72](https://github.com/omarcostahamido/PVM/commit/3cb7f72cf9ca4c46accb7b6c9b39e146cda40687) see wiki: https://github.com/omarcostahamido/PVM/wiki

You just moved the entire section there and I was suggesting only after `the part that starts with Given the objective of making this project available ... `. That is because that whole section somehow has text regarding 2 different things: the examples, and the control patch.

I like that you added images but it is confusing right now. the first image about the list of examples that `appear listed on the Max project window` show be represented by the max project window which is the slim grey window (on the left). That is the max project window.

[image]

then it mentions the #00 patch. Which is the first image you added (but please update it to show the new example 6 listed there!).

omarcostahamido, 2022-11-28:
@KaneBetter thank you for being pro-active in getting these done.
I left some comments [on the PR](https://github.com/omarcostahamido/PVM/pull/62#pullrequestreview-1195106116). Let's try to mature the readme some more before merging the updates.

KaneBetter, 2022-11-28:
I would love to help, @omarcostahamido . Since our Final Demo for the class is on December 2nd, which is also the time of hard code freeze, we still need to prepare for the public roadshow on December 6th.

omarcostahamido, 2022-11-28:
I've made several commits there, but some work is still needed.
The wiki pages also deserve a review and the missing info.

@KaneBetter good luck on your Final Demo class and public roadshow!

KaneBetter, 2022-11-30:
> * clarify what type of video files can/should be used!

Mark this done.
https://github.com/omarcostahamido/PVM/commit/11c302a67f926a66678999b06b434fad6e998f0f

KaneBetter, 2022-11-30:
> * review Structure table
Done.
https://github.com/omarcostahamido/PVM/commit/37067977ac4f9e9db6dbfe42536e2e750cbf67f1

KaneBetter, 2022-11-30:
> * review and complete wiki pages

Mark this done.
https://github.com/omarcostahamido/PVM/wiki/The-Control-Interface-in-MAX
https://github.com/omarcostahamido/PVM/wiki/Examples

omarcostahamido, 2022-11-30:
> > * review and complete wiki pages
> 
> Mark this done. https://github.com/omarcostahamido/PVM/wiki/The-Control-Interface-in-MAX https://github.com/omarcostahamido/PVM/wiki/Examples

Thanks for the updates on `The Control Interface in MAX`.
For the `Examples` page I was thinking more of a walkthrough using the patches, what each part/object does there. Similar to that text that I've written for `pvm/maxpat`.

omarcostahamido, 2022-12-02:
the Structure table will also need some more thought into it. Category names, which items goes where, and which ones to include or not. It will mature.

PR #62 (2023-02-18): Update readme
This fixes #57 .
closed 2023-02-18 by omarcostahamido